tscircuit/cli #3: Automatically install `.d.ts` types for imported snippets when someone runs `snippets dev ./path/to/snippets.tsx`

seveibar, 2024-12-15:
- [ ] Automatically populate `node_modules/@tsci/seveibar.red-led/index.d.ts` with the content from the snippets api (`snippet.dts`), creating directories where necessary

ShiboSoftwareDev, 2024-12-15:
/attempt #3

ShiboSoftwareDev, 2024-12-15:
@seveibar how do i access the snippets api? is is the registry-api?

seveibar, 2024-12-15:
@ShiboSoftwareDev sorry for the delay, yes it's the registry api!

seveibar, 2024-12-15:
@ShiboSoftwareDev it shouldn't be needed but there's also a config manager that let's you access the auth token from the original cli

ShiboSoftwareDev, 2024-12-15:
I injected the types, but they are not known inside the project, @seveibar 
[image]

ShiboSoftwareDev, 2024-12-15:
this is how the files look
[image]

seveibar, 2024-12-15:
@ShiboSoftwareDev hmm not sure, we might need to put in a package json file too

ShiboSoftwareDev, 2024-12-15:
@seveibar I had one but maybe it's missing something, I don't have to do bun add don't I?
 [image]

seveibar, 2024-12-15:
Maybe set the main entry to "index.js"?

seveibar, 2024-12-15:
Technically you could insert the registry .npmrc and fully install the package normally with "bun add". That is acceptable but wont fix type issues because the registry may not support returning types (or it might not sure)

seveibar, 2024-12-15:
[image]

seveibar, 2024-12-15:
That full package installation isnt a bad idea to do even if its currently broken because it'll allow for theoretical future execution. The downside is you have to call to shell and npm installation is technically insecure

seveibar, 2024-12-15:
I guess im ok with either i guess but if you do installation i would make sure the code is pretty rock solid because itll will have operating system and runtime bugs....

It might be nice to have the minimal types version...

ShiboSoftwareDev, 2024-12-15:
I'll see what I can do

ShiboSoftwareDev, 2024-12-16:
@seveibar today it worked without I do anything, maybe removing the old test files caused a bug in my editor

ShiboSoftwareDev, 2024-12-16:
@seveibar this is the content from the fetched dts:
[image]
does it look correct?

PR #8 (2024-12-17): automatically install imported snippet types
/claim #3
/closes #3
closed 2024-12-17 by ShiboSoftwareDev