tscircuit/cli #33: Add `tsci push` command to push (save) snippet code to Registry API

seveibar, 2025-01-20:
Note: You'll want to reference the snippets implementation. You just need to send up the circuit json and typescript/json source files

/bounty $15

kom-senapati, 2025-01-22:
@seveibar Its dependent on https://github.com/tscircuit/cli/pull/23

ArnavK-09, 2025-01-30:
should use /snippets/create or /packages/create
and for circuit json, do we need to implmenet worker??

ArnavK-09, 2025-01-30:
/attempt #33

imrishabh18, 2025-01-30:
@ArnavK-09 I think you will need to use the `/snippets/create` endpoint (reference from the fake-api in the tscircuit.com repo)

ArnavK-09, 2025-01-31:
> @ArnavK-09 I think you will need to use the `/snippets/create` endpoint (reference from the fake-api in the tscircuit.com repo)

These endpoints are going to be deprecated as per discord message!

PR #44 (2025-02-04): feat: tsci push snippet
/claim #33 
Fixes #33

https://github.com/tscircuit/cli/issues/33#issuecomment-2625173197
closed 2025-02-04 by seveibar