tscircuit/cli #67: On `tsci dev` seeing the default circuit always

imrishabh18, 2025-02-13:
Seeing the default circuit always on server start, irrespective of the index.tsx snippet code

Anshgrover23, 2025-02-17:
@imrishabh18 no, the thing is it is due to snippet-1.tsx file is always selected in dev script 

[image]

jasonobrien, 2025-02-17:
Is that for the `tsci dev` command, or just `npm run dev` inside of the CLI repo?

It'd certainly explain the behavior, but I don't think that command is being run for me, as I don't have `bun` installed.

I'm just following the quick start guide:

```
npm install -g tscircuits
mkdir my-project
cd my-project
tsci init
tsci dev
```

But - no matter what I do to `index.tsx` (which the CLI does watch and announce it's reloading) - the web UI shows the same circuit (a 1k resistor and 1uF capacitor).

imrishabh18, 2025-02-17:
@jasonobrien Assuming it's not a type on your end, you need to install the cli package by this command `npm install -g @tscircuit/cli` we are not using `npm install -g tscircuits` (it's an aggregator repo). 

I think maybe this issue could be because of bundling, but not sure. Somebody will have to take a deeper look

jasonobrien, 2025-02-17:
Yep was a typo. Fwiw it also doesn't seem to be a refresh issue, as the circuit it displays doesn't match the default index.tsx either (capacitor is a few orders of magnitude off).

imrishabh18, 2025-02-21:
I think this is fixed now

closed 2025-02-21 by imrishabh18