tscircuit/cli #109: Get preview with RunFrame to be full screen

seveibar, 2025-03-19:
[image]

requires a runframe fix, but i'm putting on this repo because this is where it matters

/bounty $20

MustafaMulla29, 2025-03-19:
@seveibar I think the issue is with "isFullScreen" toggle. Coz if "isFullScreen" is false(which it is by default) it's setting the height to "620px". Isn't it?

MustafaMulla29, 2025-03-19:
> [@seveibar](https://github.com/seveibar) I think the issue is with "isFullScreen" toggle. Coz if "isFullScreen" is false(which it is by default) it's setting the height to "620px", and when you click the "isFullScreen" toggle button which is on the top right corner of the header then it sets it height to be full.

@seveibar actually I think there is no use of that button if it does this👆

krushnarout, 2025-03-19:
/attempt #109

closed 2025-04-23 by imrishabh18