tscircuit/cli #154: tsci dev gives error

divanshu-go, 2025-04-12:
```bash

┌──(user㉿war)-[~]
└─$ tsci init  
A new version of tsci is available (0.1.80 → 0.1.81).
Would you like to update now? (y/n): yCreated: /home/user/index.tsx
Created: /home/user/.npmrc
Created: /home/user/package.json
Created: /home/user/tsconfig.json
Created: /home/user/.gitignore
Initializing project in /home/user...
Updated package.json to remove unnecessary fields.
Installing dependencies...

added 38 packages, and audited 39 packages in 34s

3 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
Dependencies installed successfully.
🎉 Initialization complete! Run "tsci dev" to start developing.
                                                                                                                                                                                           
┌──(user㉿war)-[~]
└─$ 
                                                                                                                                                                                           
┌──(user㉿war)-[~]
└─$ tsci dev 
Found entrypoint at: /home/user/index.tsx
Installing types for imported snippets... done
node:internal/deps/undici/undici:13502
      Error.captureStackTrace(err);
            ^

TypeError: fetch failed
    at node:internal/deps/undici/undici:13502:13
    at async function_ (file:///home/user/.bun/install/global/node_modules/@tscircuit/cli/dist/main.js:433157:22)
    at async DevServer.upsertInitialFiles (file:///home/user/.bun/install/global/node_modules/@tscircuit/cli/dist/main.js:438967:7)
    at async DevServer.start (file:///home/user/.bun/install/global/node_modules/@tscircuit/cli/dist/main.js:438935:5)
    at async Command.<anonymous> (file:///home/user/.bun/install/global/node_modules/@tscircuit/cli/dist/main.js:439052:5) {
  [cause]: Error: read ECONNRESET
      at TCP.onStreamRead (node:internal/stream_base_commons:216:20) {
    errno: -104,
    code: 'ECONNRESET',
    syscall: 'read'
  }
}

Node.js v22.14.0
                                                                                                                                                                                           
┌──(user㉿war)-[~]
└─$ 

```

divanshu-go, 2025-04-12:
@seveibar 

To reproduce the error
```bash
tsci init
tsci dev
```

seveibar, 2025-04-14:
I don't get this error fwiw, what is your OS?

ArnavK-09, 2025-04-15:
Probably due to poor network