tscircuit/cli #166: Renaming files aren't picked up properly

seveibar, 2025-04-16:
After a filesystem rename happens, RunFrame errors that the previous file couldn't be found. You can do a test where you have code like:

- notreferenced.tsx
- index.tsx

```tsx
import something from "./something2.tsx"
```

Now rename "notreferenced.tsx" to "something2.tsx". It will not be picked up (the circuit will refuse to render)

/bounty $20

ArnavK-09, 2025-04-17:
unable to reproduce, working fine for me ;/

roshanDev-eng, 2025-04-23:
/attempt #166

ArnavK-09, 2025-04-23:
> /attempt #166
> 

Btw, are you able to reproduce this issue?

ArnavK-09, 2025-04-23:
Bruh why algora added my name

roshanDev-eng, 2025-04-23:
/claim #166

roshanDev-eng, 2025-04-24:
@seveibar, @imrishabh18, @ArnavK-09
I’ve updated the pull request with the required changes. Once you review it, I will share the video proof to confirm the fix. Please check and let me know if anything else needs adjustment.

Anshgrover23, 2025-05-11:
/attempt #166

closed 2025-05-12 by imrishabh18

seveibar, 2025-08-17:
/tip @Anshgrover23 $20

PR #197 (2025-05-12): fix: Renaming files aren't picked up properly
/claim #166
Fixes #166 

## screenshots
[image]