tscircuit/cli #188: pushing pachages from cli succeeds while telling the user it failed

ShiboSoftwareDev, 2025-05-07:
I pushed a package many times trying to debug and this happened
[image]

ArnavK-09, 2025-05-07:
SAME with me , but i expiernced on both cli pushed and website pushed pkgs 

[image]

ShiboSoftwareDev, 2025-05-07:
hmm, what happened is that I ran bun start in the cli repo, and clicked push expecting to push my package under development, but it pushed the whole of the cli, lol
[image]

ShiboSoftwareDev, 2025-05-07:
@seveibar it went through all the files and just froze with no feedback

[image]

seveibar, 2025-05-07:
/bounty $18

seveibar, 2025-05-07:
CC @imrishabh18 this might be a backend issue

Apocalypse96, 2025-05-29:
/attempt #188

imrishabh18, 2025-06-09:
@Brijeshthummar02 I have not encountered this issue, most probably it's fixed. Happy to open if somebody faces this

closed 2025-06-09 by imrishabh18