tscircuit/cli #191: ui file -> import does not work properly

Anshgrover23, 2025-05-08:
[image]

[image]

Anshgrover23, 2025-05-08:
@seveibar yes.

ShiboSoftwareDev, 2025-05-08:
@Anshgrover23 it worked for me, either you have an old commit and or you haven't run bun i, or this only works on windows

[image]

ArnavK-09, 2025-05-08:
_Umm we should add required option to ask for cli version while creating issue, in issue template_

Anshgrover23, 2025-05-09:
@ShiboSoftwareDev i am using linux currently, so it might be possible that it is a OS specific issue, i am looking into it.

divanshu-go, 2025-05-10:
@ShiboSoftwareDev can u show by importing through UI. Can u show a video reference of this working.

ShiboSoftwareDev, 2025-05-10:
> [@ShiboSoftwareDev](https://github.com/ShiboSoftwareDev) can u show by importing through UI. Can u show a video reference of this working.

it works, you can just take my word for it, don't want to record a video because the install takes too long, that's another old issue

ShiboSoftwareDev, 2025-05-10:
@Anshgrover23 unix specific issue is highly likely, why didn't you investigate the issue yet? I doubt it's a difficult fix

PR #204 (2025-05-22): Fix: importing jlcpcb parts not being added to the file system
/close #191
closed 2025-05-22 by imrishabh18