tscircuit/cli #240: the snapshot command adds unwanted config files

ShiboSoftwareDev, 2025-06-19:
cc @seveibar , @techmannih 
[image]

seveibar, 2025-06-19:
Yes snapshot should not add config files!!!

ArnavK-09, 2025-06-21:
i guess its fixed now

sidd190, 2025-09-16:
This is fixed and can be closed.

I tried 3 cases, one with a simple folder with tsci init. One with nested folders to check changes and with with a simple index.circuit.tsx file. Performed multiple snapshots with and without flags and neither changed/added the config files at all. 

Adding the .circuit.tsx works just fine, So seems fixed.

[image]